riscvarchive/riscv-CMOs #16: Give CMO TG members ability to post to https://github.com/riscv/riscv-CMOs/wiki

AndyGlew, 2020-09-28:
My initial work for the CMO TG was done on a GitHub wiki, and now lives in https://github.com/riscv/riscv-CMOs/wiki

I would like other members of the CMO TG to be able to edit the wiki - on the web page, in their browser, not just by cloning, editing offline, and sending me a pull request.

NOT globally readable.  Ideally just CMO TG members, but RISC-V mermbers more broadly okay.

Q: what is the best way to do this?  I.e. what is the way that RISC-V higher-ups prefer?

--

I would prefer not to spend a lot of time adding new members as they sign up.

AFAICT GitHub accounts will be required for such access. That's okay. Many members do not have GitHub accounts, but most can get them.

--

Perhaps we should just tell all CMO TG members to 

a) get GitHub accounts

and ...

b) join the GitHub RISC-V organization https://github.com/orgs/riscv/people

and 

c) join a GitHub team for the CMOs TG 

which it looks easy enough to create - https://github.com/orgs/riscv/teams -- https://github.com/orgs/riscv/new-team - probably call it CMOs-members, to be consistent with the existing team CMOs-admins

I don't think that GitHub projects are relevant here.

--

I created this issue when I thought that I might require Stephano Cetola or other RISC-V higher ups to do it for me - but it looks like I can at least start the ball rolling using GitHub's request mechanism,.

Holding off, however, to give Stephano or others a chance to say

e.g. "don't bother creating a CMOs-members group - just let anyone who is a RISC-V GitHub organization member join."

TIMING: posting 9-28-2020 1pm --- will wait a day or so, for anyone to get back to me, and if no response, just do it as I describe above.

BTW, Stephano - please do NOT just do this yourself - if creating the CMOs-members group as above is okay, tell me, so that I can learn how to do it myself - or so that we can test if the DIY approach works.

TBD: my co-chair David Kruckemeyer should be an assignee for this issue - but he does not yet have a GitHub account :-)

dkruckemyer-ventana, 2020-09-29:
> TBD: my co-chair David Kruckemeyer should be an assignee for this issue - but he does not yet have a GitHub account :-)

I do have an account, but it's not clear to me yet how to take or share ownership of an issue....

dkruckemyer-ventana, 2021-08-04:
Closed the wiki for now.

closed 2021-08-04 by dkruckemyer-ventana