riscvarchive/riscv-CMOs #18: Invited Talks - collect suggestions

AndyGlew, 2020-09-28:
We probably need to invite some outside speakers to present at the CMOs TG meetings. 

Other RISC-V TGs do.

We don't want every meeting to be a presentation,. Perhaps a ratio of circa once every 3-4 meetings?

--

This issue top page to collect such a list.   Comments for suggestions.

Note: if discussing such invites is sensitive - e.g. you don't want to hurt the feelings of a famous professor by inviting him or her, or not, private email okay.  

But public suggestions welcomed too.

AndyGlew, 2020-09-29:
Gernot Heiser - secure microkernels - Has presented an "abstract ISA"  for inference channel mitigations.

My key take away from  Gernot's presentations has been
*  to control inference channels, you must partition any shared state.
*  Spatially, or in time (cache flushes)

He has expressed willingness,  schedule pending

dkruckemyer-ventana, 2020-10-13:
Someone from the software standing committee or Platform/Profile TG to describe how Linux (and other rich OSes?) provide architecturally independent interfaces to CMOs.

strikerdw, 2020-10-14:
I think we should look at both how OSes provide interfaces, but also how they actually use them in practice and what semantics they functions in those interfaces provide. So while I agree with you David, I'd be cautious personally (whatever way it falls) to just look at the interface and draw too much of a conclusion. Also, those interfaces have necessarily been bounded by what the underlying ISA provided to the OS designers. 

So, I'd hope for more than just "this is hte interface my XYZ OS has" when we get into that discussion. Also want the actual semantics provided and how things are used and why by programs on that OS and the OS itself. 

Derek

dkruckemyer-ventana, 2020-10-14:
My notion of interfaces includes semantics, so yes, I was expecting a full discussion from a list of calls to what is the expected result of each call. Usage models are also key.

dkruckemyer-ventana, 2022-01-05:
Closing due to lack of discussion.

closed 2022-01-05 by dkruckemyer-ventana