riscvarchive/riscv-CMOs #57: CMOs and PBMT

ingallsj, 2023-02-07:
Should CMOs (Clean/Flush/Inval) to memory that is PBMT-downgraded to NC or IO still search and manage the caches?
Other contemporary ISAs say that yes, CMOs to operate on caches regardless of the PTE memory type, so there is precedent for what existing software device drivers will expect.
I think we should re-open this question for RISC-V, and at least discuss and document the answer.

We do say for CBO.ZERO:
> Cache-block zero instructions store zeros independently of whether data from the underlying memory locations are cacheable.

So should we also say something like this for CBO.CLEAN/FLUSH/INVAL?
> Cache-block management instructions operate on the cache block independently of whether Page-Based Memory Types (PBMT) downgrade the memory attribute to non-cacheable.

ingallsj, 2023-02-09:
The RISC-V Privileged ISA spec says that

> Accessing the same location using different cacheability attributes may cause loss of coherence. Executing the following sequence between such accesses prevents both loss of coherence and loss of memory ordering: `fence iorw, iorw`, followed by `cbo.flush` to an address of that location, followed by a `fence iorw, iorw`.

It makes no requirement on the PBMT Cacheability of that address, therefore we can conclude that `cbo.flush` must operate on that location ignoring PBMT overrides.

dkruckemyer-ventana, 2023-02-13:
I agree, and the key idea is that CMOs ultimately respect the cacheability PMA, even if the PBMTs "downgrade" cacheability to non-cacheable.

closed 2024-05-24 by dkruckemyer-ventana