riscvarchive/riscv-CMOs #60: Clarification on CMO's interaction with PMP

deepsrc, 2023-04-17:
I need clarification on how the CMO instructions are expected to behave in this particular scenario.

The CMO spec says the following -
> The PMP access control bits shall be the same for all physical addresses in the cache block, and if write
permission is granted by the PMP access control bits, read permission shall also be granted

while the PMP specification mandates that -
> The matching PMP entry must match all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits.

Assuming grain size of 4 bytes and a cache line of 64 bytes divided into 2 contiguous PMP regions of 32 bytes each with same access permissions (say L=0 R=1 W=1 X=1), will a CBO.FLUSH result in an access fault exception?

gfavor, 2023-04-17:
For CMO's to be at all usable on a given block address (i.e. to not always get an access fault), the PMP access control bits need to be as described.  Conversely, if they are as you describe, then an access fault will always result (unless a higher priority exception occurred).

dkruckemyer-ventana, 2024-05-24:
Though it can be inferred, I don't believe the spec explicitly states that the "access size" of a CBO is the size of the block. Do folks feel that a clarification is warranted?

gfavor, 2024-05-28:
To avoid any uncertainty at the expense of just a few extra words, it would probably be good to make "access size" explicit.

closed 2024-05-28 by dkruckemyer-ventana

Timmmm, 2024-05-29:
> Assuming grain size of 4 bytes and a cache line of 64 bytes divided into 2 contiguous PMP regions of 32 bytes each with same access permissions (say L=0 R=1 W=1 X=1), will a CBO.FLUSH result in an access fault exception?

See #1313

The answer is that

1. When the PMP spec is talking about "memory accesses" it should really say "memory operations" (I am working on a PR to fix this). PMP checks are performed on memory operations, not memory accesses.
2. A `cbo.flush` can optionally be decomposed into more than one memory operation.

If the `cbo.flush` is performed as a single memory operation then it must fail. If it is decomposed into e.g. two 32 byte memory operations then it must succeed.

So either could happen. I'm not exactly sure what @gfavor was saying but hopefully the same thing!