riscvarchive/riscv-CMOs #62: Is a cache-block management instruction permitted to access device region?

YenHaoChen, 2023-06-08:
Hello,

Can the cache-block management instruction access a device region with load permission without raising an exception?

[image]

Thank you.

Best regards,
YenHaoChen

gfavor, 2023-06-10:
This warrants clarification in the spec, but in the meantime ...

**Can the cache-block management instruction access a device region with load permission without raising an exception?**

By default yes.  Also note that the Svpbmt extension says:

> Accessing the same location using different cacheability attributes may cause loss of coherence.
Executing the following sequence between such accesses prevents both loss of coherence and loss
of memory ordering: fence iorw, iorw, followed by cbo.flush to an address of that location,
followed by a fence iorw, iorw.

On the one hand, noncacheable accesses may not look up in, update, or invalidate any caches.  On the other hand a cbo maintenance instruction must be able to successfully execute with expected functionality on caches so as to be able to reliably re-establish coherence between cacheable and noncacheable accesses to the same address.

In the degenerate case that an address is only accessed noncacheably, then there is never a need for cache maintenance.  But a maintenance instruction must still execute successfully (to allow for the preceding scenarios).  In essence, the cbo executes like in the preceding paragraph.  But since the address is guaranteed to not be in any caches, the cbo won't finding anything to actually do in any caches - effectively degenerating into a nop in behavior.

Lastly, to be complete, an address may switch from being accessed cacheably to being access only non-cacheably.  A maintenance cbo would be used during that transition and should work as expected even though, from a post-transition perspective, the address is purely a noncacheable device address.

**Or does the cache-block management instruction raises access fault on the noncacheable region?**

The exception to all the above is if a system chooses to implement a "cbom" PMA attribute that separately controls whether maintenance cbo's can be performed to a PMA region.  In which case one could choose to disallow cbom accesses to some or all PMA I/O regions (resulting in an Access Fault).  Or an implementation could choose to simply disallow maintenance cbo's to all PMA I/O regions based on the "I/O" attribute.  (Note that Svpbmt cannot make a PMA "I/O" region accessible in a cacheable manner.)

YenHaoChen, 2023-06-12:
Thank you for the detailed clarification.

YenHaoChen, 2023-06-12:
Another follow-up question. 

What happens when a cbo.zero accesses a device region with load permission? The software developer may not expect the NOP behavior in this case. Does raising an exception make more sense?

(BTW, I think prefetch instructions work as NOP well on device region.)

closed 2023-07-10 by YenHaoChen