riscvarchive/riscv-CMOs #67: Must cbo.clean, cbo.flush, and cbo.inval raise exception on an address in vacant region?

YenHaoChen, 2024-04-17:
Hello,

I want to clarify the permissions for cache-block management instructions. Consider a vacant region that PMP/PMA permits access to. Does any cache-block management instruction raise an exception on a cache block in that vacant region? Or should an implementation treat the instruction as nop in this case?

[image]

ingallsj, 2024-04-17:
> Consider a vacant region that PMP/PMA permits access to.

A vacant region is defined in the Privileged ISA spec as PMA **not** supporting access to.

> Vacant regions are also classified as I/O regions but with attributes specifying that **no accesses are supported**.

YenHaoChen, 2024-04-17:
Thank you for the clarification. Therefore, for an incorrect PMA configuration that allows access to vacant regions—whether due to a buggy software setting or hardware implementation—we indeed classify it as unspecified behavior when accessing addresses within these regions. Is this understanding accurate?

ingallsj, 2024-04-17:
Buggy behavior is unspecified in the ISA.

closed 2024-04-17 by YenHaoChen