riscvarchive/riscv-CMOs #68: prefetch i/r/w encoding overlaps with ORI instr

akifejaz, 2024-10-20:
Hey! I was reading out this spec last week and I was trying to implement it on my processor but I'm facing some issues related to decoding of instructions. I've already hooked up RV & RVV base ISA.

As per the ORI instruction, its encoding is like 
[image]

and prefetch.i has encoding
 
[image]

Since the opcode, func3 are same, only distinguish thing I was expecting the func6 of prefetch, but seems like if we have ORI instr with let say 0-3 value as imm while decoding it will become hard to distinguish if the imm is for ORI or its func6 of prefetch. 

can someone please clear me on this? Thanks!

gfavor, 2024-10-20:
The prefetch instructions use HINT encodings reserved within the ORI encoding space.  Look at the spec for HINT instructions.

akifejaz, 2024-10-21:
> The prefetch instructions use HINT encodings reserved within the ORI encoding space. Look at the spec for HINT instructions.

Thanks for response, so if I'm not wrong, for prefetch instr the `'offset[4:0]` or `rd` will be zero always.

gfavor, 2024-10-21:
Yes.