riscvarchive/riscv-CMOs #71: Clarification needed: senvcfg check in systems without S-mode

mimiqdev, 2025-01-21:
Hi,

In the CMO specification, the pseudo code for CBO.FLUSH instruction includes checking senvcfg.CBCFE when in U-mode:

```
// illegal instruction exceptions
if (((priv_mode != M) && !menvcfg.CBCFE) ||
((priv_mode == U) && !senvcfg.CBCFE))
{
<raise illegal instruction exception>
}
// virtual instruction exceptions
else if (((priv_mode == VS) && !henvcfg.CBCFE) ||
((priv_mode == VU) && !(henvcfg.CBCFE && senvcfg.CBCFE)))
{
<raise virtual instruction exception>
}
// execute instruction
else
{
<execute CBO.CLEAN or CBO.FLUSH>
}
```

Question: How should this check be handled in systems that don't implement S-mode? Should we:
1. Skip the senvcfg.CBCFE check entirely in such systems?
2. Treat it as if senvcfg.CBCFE == 0?
3. Only check menvcfg.CBCFE?

Similar questions for senvcfg.CBIE/CBZE as well

gfavor, 2025-01-21:
If S-mode doesn't exist, then senvcfg doesn't exist either.  Which leaves menvcfg as the only controlling envcfg CSR.

mimiqdev, 2025-01-21:
Thanks for the clarification, I created a related issue in spike.

closed 2025-01-21 by gfavor