riscvarchive/riscv-CMOs #72: The value of mCBIE is 0,resulting in cbo.inval an illegal instruction.

soccor-crypto, 2025-04-11:
When the cbo.inval instruction is involved, an instruction exception error occurs, and it is found that the value of sCBIE mCBIE hCBIE is 0. May I ask how to configure the value of CBIE so that the value of sCBIE mCBIE hCBIE is 1? Or is there any other way to implement cbo.inval as a valid directive?

The median value is as follows, because value of STATE.menvcfg->read() is 0, resulting in the value of mCBIE being 0
CSR_MENVCFG: 30a
menvcfg_mask: f0
menvcfg: 0x1dbcf 40
STATE.menvcfg->read(): 0
MENVCFG_CBIE: 30

gfavor, 2025-04-11:
The various CBIE bits are in the *envcfg CSRs.  Architecturally speaking, one simply writes those bits using CSR instructions.  Past that, it is unclear what is being asked?

soccor-crypto, 2025-04-11:
> The various CBIE bits are in the *envcfg CSRs. Architecturally speaking, one simply writes those bits using CSR instructions. Past that, it is unclear what is being asked?

Thank you for your answer. The log indicates menvcfg: 0x1dbcf 40, but the actual read value is STATE.menvcfg->read(): 0. So is it possible that something needs to be set up here?

soccor-crypto, 2025-04-11:
> The various CBIE bits are in the *envcfg CSRs. Architecturally speaking, one simply writes those bits using CSR instructions. Past that, it is unclear what is being asked?
Spike defaults to STATE.menvcfg->read(): 0, so I need to configure this register myself, right? But as the source code,
CSR_MENVCFG: 30a
menvcfg_mask: f0
menvcfg: 0x1dbcf 40
MENVCFG_CBIE: 30
 it looks like spike's default value is not expected to be TATE.menvcfg->read(): 0.

soccor-crypto, 2025-04-11:
> Thank you for your answer. The log indicates menvcfg: 0x1dbcf 40, but the actual read value is STATE.menvcfg->read(): 0. So is it possible that something needs to be set up here?

Spike defaults to STATE.menvcfg->read(): 0, so I need to configure this register myself, right? But as the source code,
CSR_MENVCFG: 30a
menvcfg_mask: f0
menvcfg: 0x1dbcf 40
MENVCFG_CBIE: 30
it looks like spike's default value is not expected to be TATE.menvcfg->read(): 0.

gfavor, 2025-04-11:
Since this a Spike issue (versus an ISA manual issue), there will be better prospects of getting an answer by posting this question against the riscv-isa-sim github repo.

soccor-crypto, 2025-04-11:
Ok, thank you. I'll go to riscv-isa-sim github repo for some advice.

closed 2025-04-11 by soccor-crypto